byuccl/spydrnet #32: Implement and Test Callback Framework

andrewmkeller, 2020-01-06:
We want to add a function call anytime that an object is added, removed, or has its properties modified.

Tests for this task would include things like registering a call back and making sure it gets called when the event happens.

Documentation is also needed.

We talked about having a refresh or update function call so that call backs can be disabled temporarily for major changes.

Add any related comments or updates to this issue.

thunder-hammer, 2020-01-08:
list of functions to have callbacks on list subject to change:
Element.__setitem__
Element.__delitem__
Element.pop
Cable.add_wire
Cable.remove_wire
Definition.add_port
Definition.remove_port
Definition.add_child
Definition.remove_child
Definition.add_cable
Definition.remove_cable
Instance.reference (setter)
Library.add_definition
Library.remove_definition
Netlist.top_instance (setter)
Netlist.add_library
Netlist.remove_library
Port.add_pin
Port.remove_pin
Wire.connect_pin
Wire.disconnect_pin

andrewmkeller, 2020-01-08:
I like it. This list feels complete. At some point we may want to add:

sdn.add_netlist
sdn.remove_netlist

These functions are for adding the netlist object to the global environment (they don't exist yet). Basically before a netlist is added to the environment it is unmanaged, but when it is added to the environment then it is managed. You could imagin the EDIF data manager listening for two netlists to be added and checking for name clash between the two netlists. This is the only use case that I think would be netlist independent.

thunder-hammer, 2020-01-08:
That makes sense. I will start with the netlist dependent features and the independent ones should not be too hard from there.

thunder-hammer, 2020-01-10:
I think I have a good framework of callbacks going on. I don't think this is ready for release yet though, it ended up taking me a little more time to figure out a good way to do this. I'll put some time in over the weekend and these should be in a much better place Monday. I also think that we can use the datamanager as an example of how to use the callbacks.

What slowed me down most was probably that I couldn't find a good way to consolidate callback functions without enumerated types or some really crazy string parsing on something like string(method) calls. I decided to simply not consolidate them.

closed 2020-01-18 by andrewmkeller