byuccl/spydrnet #33: Implement and Test EDIF Namespace Data Mgr

andrewmkeller, 2020-01-06:
This is a plug-in that uses the callback framework to manage the edif namespace.

Here is the namespace:

Any element (except for pins and wires at this point) can contain metadata / be named.

EDIF.original_identifier (or EDIF.identifier if no original identifier exists) will be stored in the ['NAME'] property. The case sensitive version of the EDIF.identifier will be stored in the ['EDIF.identifier'].

netlists have their own namespace (two different netlists by the same case-sensitive name, or the same case-insensitive EDIF.identifier) cannot co-exist (that namespace obviously is at the very top of the SDN global manager since it applies to multiple netlists.)

There is a namespace within a netlist for libraries, and a namespace within a library for definitions. Definitions have three separate namespaces ports, cables, and instances respectively.

Testing would include making sure thing clash when they are suppose to clash and not when they shouldn't.

Additional features of this manager would also include generating unique names from a seed, or checking names for uniqueness, etc.

thunder-hammer, 2020-01-21:
I think complete testing of this will resolve issue #65

thunder-hammer, 2020-02-07:
This should be complete, there is currently a pull request #66 that will resolve this issue. Unique names however are not implemented in the current data manager. I think we could move that into a new project though if we wanted.

thunder-hammer, 2020-02-07:
Netlist level watching is also not yet implemented.

thunder-hammer, 2021-05-17:
I am going to close this issue. the callback framework seems to be complete enough for now. New issues could be open for additional work with the callbacks.

closed 2021-05-17 by thunder-hammer