byuccl/spydrnet #54: Verilog to EDIF attempt

andrewmkeller, 2020-01-20:
Very similar to issue #53. Just trying to push a verilog parsed netlist through the edif composer. What steps need taken to "convert" or prepare or pre-process the netlist so that it can be correctly output as EDIF? What we find out from this task will become a list of things we need to do the make this conversion more clean and possible.

thunder-hammer, 2020-09-02:
Now that we have the parser and composer done. I plan on tackling this.

There are going to be several conversion features to assess.

 * Verilog has assign statements. -may need to adjust the verilog parser and composer to better handle this.
 * Verilog has multi wire cables and multi pin ports. -edif composer
 * Verilog's naming convention is less restrictive. -I will try and take a look at how to best handle this. issue #116 seems related and may be fixable at the same time. I will comment here and there accordingly
 * Verilog has some port reassignment capabilities that I don't think are in EDIF.

And of course there will be things that I'm not anticipating now.

thunder-hammer, 2021-05-17:
The EDIF Composer should be able to support this to some extent.

some things may need additional testing and improvement
do verilog assignment statments work propertly
Add the ability to translate parameters from verilog