byuccl/spydrnet #55: Verilog Parser

andrewmkeller, 2020-01-20:
Take a structural netlist (generated by vivado for now) and parse it into our netlist format. This will likely involve creating a tokenizer and a resursive decent parser similar to what we have done with EDIF. Feel free to explore other options, but remember that to begin with, we want a composer that works more than the best solution that might work down the road.

andrewmkeller, 2020-01-23:
Here are some excellent resources for looking at this: https://tomassetti.me/parsing-in-python/ https://tomassetti.me/guide-parsing-algorithms-terminology/

There are many ways to do this... Ultimately, having a working parser and composer is best, but if we could take advantage of PLY or another framework, it may be really easy to crank out more and more languages.

thunder-hammer, 2020-04-30:
This has been worked on extensively by blazing_marimba. I am picking up the work where he left of. I'll keep this ticket updated as I go.

thunder-hammer, 2020-05-01:
The parser is currently using ply. I was doing some research and this is no longer updated as a pip repository (according to the website) I still think that it would be beneficial to leverage another framework. I was also looking at converting the code over to use sly which is supposedly more modern.

ply

pros:
 - well tested and been around for a long time.
 - still receives bug fixes and the API likely won't change much on us
cons:
 - latest versions are not pip installable (version 3.11 was the last might be good enough)

sly

pros:
 - more modernized (not entierly sure this is a pro haha)
 - more likely to have updated pip version (0.4 latest version)
cons:
 - doesn't support python version 3.5
 - currently has a disclaimer that it might not work (alpha/beta software)

do it ourselves

pros:
 - fewer dependencies
 - might be faster?
cons:
 - we have to maintain it
 - we are the only ones looking for bugs
 - might take longer?

thunder-hammer, 2020-05-06:
We made the decision to create our own parser.

I am working on re-purposing the edif tokenizer to tokenize verilog.

thunder-hammer, 2020-09-02:
This has been released in 1.5.0.

closed 2020-09-02 by thunder-hammer