byuccl/spydrnet #60: Cloning and deep copy of elements from whole netlist to smallest element.

andrewmkeller, 2020-01-20:


thunder-hammer, 2020-01-28:
created a branch deep_copy for this issue. forgot to mention it though in the commit...

thunder-hammer, 2020-01-28:
Looking at the python built in copy module I think this might most easily be accomplished by overriding the python __deepcopy__() function for each class. I'll take a look at persuing this although we certainly want our own api endpoint.

thunder-hammer, 2020-01-31:
I am going to leave a brain dump on copying in this comment. I hope some of this will make it into the documentation but I just needed somewhere to write my thoughts down.

It seems straight forward to copy but it is actually rather complex. sdn.copy(element) needs to return a new element that is a complete copy of the existing element. for a netlist it needs all new libraries for a library all new definitions for definitions, ports, cables, and instances... however there is complexity in how the new copies deal with the pointers that they have to the existing netlist.

each element when copied will need to maintain itself in a dictionary that maps old item to new item.

This is an outline of how each copy function will behave

OuterPin
 * copied but still connected to the wire the old one was (to allow upper levels to copy it correctly)
 * connected to the same inner pin (to allow upper levels to copy it correctly)
 * orphaned from the instance

InnerPin
 * copied but still connected to the wire the old one was (to allow upper levels proper copy capability)
 * connected to the same outter pin (to allow upper levels to copy it correctly)
 * orphaned from the port it used to belong to

Wire
 * orphaned from the cable
 * still connected to the pins it used to connect but the pins are not connected to it (to allow upper levels to do correct copying)

Instance
 * reference is left in tact (shallow copy)
 * Orphaned from parent
 * all outer pins are copied but remain connected to the wires they used to be connected to (so that I can change them when copying an instance while copying the dictionary)

port
 * orphaned from parent
 * all inner pins are copied and remain connected as if copied alone
 * inner pins now belong to this port as parent in same order as old copy

cable
 * Orphaned from parent
 * all wires are copied following rules for wire copy above
 * wires are assigned to this cable as parent in same order as old copy

definition
 * creates a copy of each instance, cable, and port as listed above
 * reach into cables and change each reference to a pin to reference the new pin
 * reach into ports and change each reference to a wire to reference the new wires
 * reach into instances and change each reference to wires to reference the new wires
 * reach into instances and change each reference to inner pins to reference the new ones

Libraries
 * copy each definition following the rules for a definition copy
 * reach into the instances and change the reference values to the new copies where they exist

Netlist
 * copy each library following the rules above
 * reach into each instance and change the reference values to the new copies where they exist

thunder-hammer, 2020-01-31:
@andrewmkeller if you have any thoughts on this let me know before the release

If we leave the copies as I outlined above

The wires, ports, pins, cables, instances have strange behavior when copied. They are no longer a part of the definition, cable, port (I figure the user should decide where to put them) but they maintain connections to the definition, port, cable...

Wires are especially strange because they remain connected to pins but the pins can only connect to one wire and they are not changed.

I have a feeling if we leave this behavior and give the user access it will cause some difficult bugs to show up.

I think some of our options are:
 * present a more logical copy function to the user
 * leave them as is and document it well so the users can figure out what is happening
 * disallow copies on the elements that make the least sense. (I don't think this is the best option)

thunder-hammer, 2020-02-13:
I restructured the code. we no longer depend on pythons deep copy for the elements. However we still do use it for the built in classes. I'm currently writing more tests

closed 2020-02-13 by thunder-hammer