byuccl/spydrnet #62: Upload source for ExampleCircuit.svg and include the figure somewhere in the documentation

andrewmkeller, 2020-01-20:


thunder-hammer, 2020-01-21:
I created the image with inkscape. I'm not sure there is any other source. I did pull the circuit elements off of wikimedia commons they were licensed under CC0 or the public domain I don't recall which. I could upload those individually if it would be helpful.

andrewmkeller, 2020-01-21:
Can you edit a svg with inkscape or do you need a different file? I like
the image, but I want to make sure we can move things around if ever we
need to.

thunder-hammer, 2020-01-21:
Good question. I just did some research. Inkscape adds some information to the SVG to make it editable as it was when it was closed. I uploaded the one that is editable in inkscape. So I think it should be good to go.

andrewmkeller, 2020-02-01:
Actually, we haven't added this figure to the documentation anywhere...

thunder-hammer, 2020-09-02:
The figure is now available at https://byuccl.github.io/spydrnet/docs/stable/reference/api_specification.html

PR #68 (2020-02-01): Added example figure to the documentation
closes #62
closed 2020-09-02 by thunder-hammer