byuccl/spydrnet #79: Create a shortcut for setting top instance to an instance of a definition to which I don't have a handle

thunder-hammer, 2020-02-26:
It would be convenient to create a shortcut for this code :

`top = Instance()
top.reference = top_definition
netlist.top_instance = top`

by adding the following capability that will take care of the extra steps behind the scenes

`netlist.top_instance = top_definition`

thunder-hammer, 2020-04-27:
@andrewmkeller If you have any suggestions for the hrefs feel free to comment or edit the initial comment.

thunder-hammer, 2020-05-01:
Have fun with testing. I've found it super useful. Going into the weekly meetings if I can say I have good tests with 100% code coverage It really gives a good boost of confidence.

The following are some generic ideas for testing that might be useful.

make sure the code works the way it was expected to work. for example using the short cut and long way should yield the same result.

take a look at how to break the test. for example make sure assertions are thrown if a wire does not belong to a cable or an instance does not have a reference. If there is a way to make sure the error message makes some sense that could be really helpful down the line too.

Find other strange but good use cases, and potential abuses and find a way to set up those circumstances to see if the code still holds up.

thunder-hammer, 2020-05-12:
There are a few more shortcuts which would be useful

Basically the same shortcuts for the first class elements constructors from above would be useful to add to the parent's children constructors for example

`library.create_definition(name = "my_def")`
`definition.create_port(width = 32, downto = True)`

and so on for all of the first class element .create_* calls in the api to mirror the __init__ functions.

thunder-hammer, 2021-05-17:
@jordi1215 I think you implemented most of this, could you update this issue please?

jacobdbrown4, 2021-08-11:
@jordi1215 has implemented these (at least the ones deemed useful/necessary).

closed 2021-08-11 by jacobdbrown4