byuccl/spydrnet #102: Instances may not be unique even if all hierarchical instances are

andrewmkeller, 2020-03-31:
Imagine that the top instance was of a definition that is instanced elsewhere in the netlist. According to how things are done in EDIF it is not possible for a sub instance of a definition to also be the top instance of the netlist, however the top instance could be a definition that is instanced elsewhere in the design which would make the instances no longer unique.

thunder-hammer, 2020-04-17:
This was a problem I ran into while uniquifying too (although I think the uniquifier might make the top unique??? I'll have to check). Even if the top instance was unique a lower instance might not be. is that still a part of the problem?

One work around could be to check while composing to ensure that the instances follow the rules.

Another option or perhaps part of the first one would be to make an instance level uniquifier that could be applied to things like the top instance that need it.

Finally we could just check if the instance is top before instancing it elsewhere and disallow it but that seems like it would remove some functionality like a change top after reading in a netlist

thunder-hammer, 2020-05-18:
Should be identified with issue #108