byuccl/spydrnet #115: can we read edn file ?

WrathofBhuvan11, 2020-07-06:
sometimes xilinx exports edn file instead of edf can we read it ?

AEW2015, 2020-07-06:
Its the same file format, just a different file type name.
I usually change the file type to edf when specifying the file name.

andrewmkeller, 2020-07-06:
@AEW2015 pointed out that by default Vivado exports EDIF with a ".edn" suffix. The SpyDrNet `.parse` function currently recognizes file formats by the suffix, so if you change the suffix from ".edn" to ".edf", SpyDrNet will treat it as a EDIF netlist. SpyDrNet will be able to parse it in unless it is encrypted.

thunder-hammer, 2020-07-06:
We could add the ability to automatically parse the EDN format files through our EDF parser.

We can try have the EDF parser parse EDN files but we would like to identify encrypted files in order to give the users some useful error statements.

I think this should be doable by looking for the xilinx header and then checking to see if the first line of the file makes sense. if not we could print an error that says something along the lines of "syntax error {what was expected} expected line 1, Is this an encrypted file?"

As a work around users could change the file name at the user level then parse the changed file, do the modifications, and then change the file name back all in the python script.

closed 2020-07-12 by WrathofBhuvan11