byuccl/spydrnet #116: Parsing is wrong when we have case sensitive wire names

shervinrs, 2020-07-21:
I am using sdn.parse() for parsing yosys edf outputs. Lots of circuits might have case sensitive wire names like 'n4139' and 'N4139'. In these cases, sdn.parse() merges these wires together which is wrong. The issue seems to be related to the no_conflict() in spydrnet/plugins/namespace_manager/edif_namespace.py.

andrewmkeller, 2020-07-21:
Are 'n4139' and 'N4139' the EDIF identifiers or are they included in an EDIF "rename"? Can you share the EDIF declarations of these nets? Name clashes are dependent on both the EDIF identifier and potentially the "rename identifier" if it exists.

The EDIF parser in Xilinx Vivado merges EDIF nets under certain conditions of naming conflict. We tried to match its behavior when we wrote the parser, but we may have gotten it wrong. 

Depending on what is happening, we may need to have a separate parser for Vivado generated EDIF verses Yosys generated EDIF.

shervinrs, 2020-07-22:
I think they are identifiers. This is a part of the Yosys output:
          (net N4139 (joined
              (portRef DIN (instanceRef id00022))
              (portRef Q (instanceRef U4835))
            )
          )
          (net n4139 (joined
              (portRef DIN2 (instanceRef U3808))
              (portRef Q (instanceRef U3810))
            )
          )
But it changes to this if I write out the EDIF file again using sdn.compose().
            )(net n4139 (joined
              (portref DIN2 (instanceref U3808))
              (portref Q (instanceref U3810))
              (portref DIN (instanceref id00022))
              (portref Q (instanceref U4835))              
              )
I have both 'n4139' and 'N4139' in wire deceleration section of my Verilog file.

andrewmkeller, 2020-07-22:
Thanks for the quick response.

In EDIF 2 0 0, EDIF identifier themselves are technically case insensitive. The "Electronic Design Interchange Format Version 2 0 0 Recommended Standard EIA-548" states that: "Case is not significant in keywords, symbolic constants, or identifiers." (Page 1-2). 

If the nets are declared in the same cell view, the EDIF generated by Yosys is technically illegal. The standard says, "It is illegal for two objects of the same class [net, instance, etc.] to have the same name [case insensitive] if they are in the same scope [e.g., within the same cell view]." on page 1-4.

But Vivado treats EDIF identifiers as case sensitive when comparing them to "rename" identifiers, and Vivado will merge nets that conflict in naming under certain circumstances. We will need to run some test to verify the exact conditions.

It sounds like the expected behavior from the Yosys point of view would be to have the parser create two separate nets and not merge them?

Do you have any thoughts on how we should best proceed? The standard is from 1988 and level 2 0 0 hasn't been updated since to my knowledge. I will reach out to the Yosys team to see what they think.

andrewmkeller, 2020-07-22:
This issue is talked about in https://www.xilinx.com/support/answers/54283.html.

shervinrs, 2020-07-22:
Yes, Yosys treats these two nets as two separate wires. It seems they followed case sensitivity for EDIF similar to their Verilog parser.
I think you can print a message that you have merged these nets together or consider adding an option to the parser to switch between case sensitive and case insensitive.

thunder-hammer, 2021-02-02:
Just to keep everyone updated, this is in the backlog.  Supporting case sensitive names (maybe with a switch of some sort) would be quite valuable. I'm not sure when we will get to it but it is something we are looking at.