byuccl/spydrnet #129: Add some additional information to the documentation

thunder-hammer, 2021-04-30:
Clarify the difference between children and references in the definition
We have the instance object defined but it may be confusing what children and references are in the context of a definition

Add some documentation about the EDIF.properties and Verilog.properties dictionaries and how those work.
Additional information about the metadata would be good including which objects have it and it's intended purpose (as well as all of its potential)

jacobdbrown4, 2021-05-18:
I have added some more explanation about parents, children, and references under the page located at Reference/API Summary/Instance. I thought this would be the best place because these three things come up with instances.

jacobdbrown4, 2023-04-26:
The EDIF.properites and Verilog.properties need documentation. In reality, they need to not be seperate things but rather be a single properties dict. But for now some documentation for these would be nice.