byuccl/spydrnet #139: instances.get_pins() vs instances.pins

jacobdbrown4, 2021-05-24:
This is probably a minor issue, but right now, instance.get_pins() returns inner pins but instance.pins returns outer pins. Shouldn't they both return outer pins?

thunder-hammer, 2021-05-25:
good catch. I believe that the get_pins() call has the option to return either inner or outer pins and defaults to inner. i'll do a little bit of digging in the code and let you know what I find.

thunder-hammer, 2021-05-25:
Could you try something like the following to see what happens?

`from spydrnet.util.selection import Selection`

and then run something like

`get_pins(selection = Selection.OUTSIDE)`

Let me know if that gets the outer pins

I think we made selection = Selection.INSIDE the default on everything. The instance's pins case might be one where it doesn't make as much sense. I think we were thinking if you wanted to get all cables attached to an instance or definition, you might be interested in the ones that are inside of it or outside. We just extended that to the pins.

jacobdbrown4, 2021-05-25:
Using the Selection.OUTSIDE does get the outer pins. That makes sense now why you did that...so you could get either the inner or outer pins. Now that I understand it, it doesn't seem like a bug really so we probably don't need to change it.

agg23, 2023-04-01:
I'm new to this codebase, but I don't view this as a minor issue. The existing IR construction is already somewhat opaque given the similar names and concepts (`InnerPin`, `OuterPin`, `Port` all represent similar, but different concepts, but have little differentiation in the names), and this just adds further confusion.

I spent several hours scouring the docs and playing with the code, trying to figure out why I couldn't find the wires for a given instance, only to find out that on many of the types, `get_*` does not return the same same data as a getter for the local property.

* If `Instance.get_pins()` is not remotely equivalent to `Instance.pins`, why are they named nearly the same? As a user, unless instructed otherwise, I would expect either of those calls to give me all relevant pins for that instance. That means the `InnerPins` from the original `Definition`, and the `OuterPins` being used for that particular `Instance`. Except neither of those actually do that.
* Why is a default chosen for `get_pins()` that _filters_ data (you should basically never do this unless there are performance concerns)? 
  * If a method filters data by default, why is that not very explicitly called out in the documentation?
  * Why does `get_pins()` not have a required filter argument? I imagine you probably wanted to avoid a breaking API signature change, but I would expect that filtering by default was a change anyway.
* How do the `get*` methods differ from the local getters, what is their intended use, and where is that documented?

On a similar, but unrelated note, why does the `Selection` enum declare both `BOTH` and `ALL`? There appears to be some differentiation, given their uses in the codebase, but that is unclear from the naming.

jgoeders, 2023-09-14:
I agree with @agg23.  I wasted a good amount of time on `.pins` vs `.get_pins()`, trying to figure out why `.get_pins` on my `Instance` was not giving me the `OuterPins`.  

I don't think it makes sense to have getter and a property that sound to do the same thing.  It sounds like `get_pins()` should be renamed to something else...

jacobdbrown4, 2023-09-14:
Thanks for the feedback. This should definitely be improved. I'll take a look at it soon.