byuccl/spydrnet #141: Verilog Parser features required to parse Quartus Prime .vo/vqm files

thunder-hammer, 2021-05-25:
Currently I only have one example of each of these formats. There could be a lot more constructs than what I see.

VO files
----------
defparam structure
`defparam \avs_s1_readdata[12]~output .bus_hold = "false";`

constants mapped to ports in instances
`cyclonev_io_obuf \avs_s1_readdata[16]~output (
	.seriesterminationcontrol(16'b0000000000000000));`

VQM files
-------------

Everything needed for VO files plus

inverters on assignment statements
`assign \fp_pow_0002:fp_pow_inst|__ALT_INV__prodXY_uid496_pT2_uid435_expPolyEval_cma_s[0][49]  = ~ \fp_pow_0002:fp_pow_inst|prodXY_uid496_pT2_uid435_expPolyEval_cma_s[0][49] ;`

thunder-hammer, 2021-05-25:
The tri1 and tri0 types can be maintained like the wire and reg types. The composer needs to be updated to keep track of all these types.

thunder-hammer, 2021-05-25:
The defparam should be able to be handled much the same way the current parser currently handles the parameters. Only difference is it may have to create a new instance to add the parameter to first?

wirthlin, 2021-06-07:
Note from Andrew:

VQM files are produced by synopsis simplify. I just forward you dr Wirthlin the b13 in vqm.

thunder-hammer, 2021-06-16:
Additional constructs found in the .vo format edited into the original issue.