byuccl/spydrnet #144: Create Xilinx library

wirthlin, 2021-06-07:
Represent Xilinx built in primitives and provide functions for more easily dealing with Xilinx-specific primitives. 

- Give the library ahead of time to the parser so that the parser checks against the library (warning if the primitive is not in the library)
- Ability to generate a condensed library of only those primitives you use
- Could write a script that converts the existing libraries from VHDL/Verilog into SpyDrNet python code
- Provide more "meaning" to libraries and functions to extend that meaning in an API
- 
Example uses:
- In TMR, you could more easliy find all the Xilinx flip-flops

benglines, 2021-08-02:
SpyDrNet-SHREC repo has this functioning already. The code from there should be migrated to SpyDrNet as it isn't specific to TMR and can be useful for other purposes

jacobdbrown4, 2022-06-06:
The SpyDrNet TMR/SHREC functionality of this is different. That is for knowing types of primitives such as which ones are flip flops. I suppose these two functionalities may be able to be merged but I don't know how important that is to us.

wirthlin, 2022-06-07:
Can you expand on your comment? I am not quite following what you are saying here. Can you explain in detail how it is currently done so I better understand?

jacobdbrown4, 2022-10-19:
In SpyDrNet, one can specify what architecture the netlist is for (e.g. Xilinx 7-Series) so when the netlist is parsed, port directions of each primitive are known. It's the equivalent of having `celldefine statements in every netlist. Without this, port directions (specifically in verilog netlists) are undefined.

In SpyDrNet-TMR, there are databases (actually just one right now for Xilinx 7-Series) that specify what type of primitive each primitive is i.e. FDRE is a flip flop, OBUF is a buffer, etc. This aids in locating voter insertion spots (like finding flip flops).