byuccl/spydrnet #148: EDIF parser missing multi bit cables

jacobdbrown4, 2021-07-09:
I ran into something that seems like a bug, so I wanted to let you know and ask what you think. When I parse one of my ECEN 220 projects into spydrnet, one of the wire connections isn't right.

The wire coming from count_reg[8]'s Q port isn't recognized as connected to anything in SpyDrNet. But it should be connected, as seen in the Vivado screenshot and in the edif file.

jacobdbrown4, 2021-07-09:
Interestingly, I parsed the netlist and immediately composed it, and the line in the netlist for that particular connection was missing. I'll attach a picture of both the original, Vivado-synthesized netlist and the spydrnet parsed then composed netlist.
[image]

jacobdbrown4, 2021-07-09:
From @thunder-hammer :+1: 

I was able to run this through the parser, then compose and parse the composed file and run the internal compare function. Usually that shows significant differences. It didn't say it found a difference. Makes me think it's not an issue with the composer (although we can't rule it out entirely, it could be a problem with both the composer and compare function). This makes sense with your code showing that the pin does not show up. This also means our tests are likely not capturing the issue if it happens with our test files.

interesting observations looking at the files:

The bus count_reg_n_0_[#] has the same issue for each of the indices from 1 to 9 but not index 0, and 10 and up.
The apparently same problem happens with some other nets for example
Original:
         (net (rename count_val_reg_n_0__1_ "count_val_reg_n_0_[1]") (joined
          (portref Q (instanceref count_val_reg_1_))
          (portref (member S 2) (instanceref count_val_reg_0__i_1))
          )
         )
Composed:
            )(net (rename count_val_reg_n_0__1_ "count_val_reg_n_0_[1]") (joined
              (portref Q (instanceref count_val_reg_1_))
             
              )

jacobdbrown4, 2021-07-09:
I agree that this bug is probably an EDIF parser issue. I found another good example of the problem in the built-in spydrnet example netlist 'register_file'. It has top level input ports sw[12:0]. When the netlist is parsed, only sw 0 and 10-12 are connected, while sw 1-9 are disconnected. Maybe the register_file example is easier to work with when debugging because the bug is right there on the top level. Not sure.

jacobdbrown4, 2021-07-09:
First, sorry for the super verbose email, If you skip to the bottom I think I found the exact issue.

I boiled it down to a 3 net example file.

I'll attach 2 files that illustrate. min_good will parse and compose fine. min_bad will have the issue on bit 1 of the net.

The input and output on min_bad are as follows.

Input:

         (net (rename count_reg_n_0__0_ "count_reg_n_0_[0]") (joined
          (portref I0 (instanceref count_0__i_3))
          (portref I0 (instanceref count_3__i_8))
          (portref Q (instanceref count_reg_0_))
          )
         )
        (net (rename count_reg_n_0__2_ "count_reg_n_0_[2]") (joined
          (portref I2 (instanceref count_3__i_8))
          (portref Q (instanceref count_reg_2_))
          (portref (member S 1) (instanceref count_reg_0__i_2))
          )
         )
         (net (rename count_reg_n_0__1_ "count_reg_n_0_[1]") (joined
          (portref I3 (instanceref count_3__i_8))
          (portref Q (instanceref count_reg_1_))
          (portref (member S 2) (instanceref count_reg_0__i_2))
          )
         )

Output:

(net (rename count_reg_n_0__0_ "count_reg_n_0_[0]") (joined
              (portref I0 (instanceref count_0__i_3))
              (portref I0 (instanceref count_3__i_8))
              (portref Q (instanceref count_reg_0_))
             
              )
            )(net (rename count_reg_n_0__1_ "count_reg_n_0_[1]") (joined
              (portref I3 (instanceref count_3__i_8))
              (portref (member S 2) (instanceref count_reg_0__i_2))
             
              )
            )(net (rename count_reg_n_0__2_ "count_reg_n_0_[2]") (joined
              (portref I2 (instanceref count_3__i_8))
              (portref Q (instanceref count_reg_2_))
              (portref (member S 1) (instanceref count_reg_0__i_2))
             
              )
            )

If the nets are in order in the input this is not an issue. It looks like Vivado orders nets with alphabetical ordering so it outputs the nets not in strictly increasing numerical order (_10_ comes before _2_). Since SpyDrNet composes in numerical order it does not show up when parsed from a SpyDrNet composed netlist. (making it slip past our tests)

The code in SpyDrNet that I think is the culprit is the method multibit_add_cable around line 926 in /parsers/edif/parser.py That code's purpose is to combine nets based on their names suffix for example combine net_0_ and net_1_ into a 2 bit cable. It looks for an already combined cable, and starts to add bits to it. I think it is probably related specifically to lines 956 to 963 in that file. I added some print statements so the line numbers might not be exactly on. I'm also pretty sure the cable that is passed in has 3 pins. they are just not getting added to the final cable.... haha just figured it out:

pins = w.pins
for pin in pins:
    print(pin)
    w.disconnect_pin(pin)
    ew.connect_pin(pin)

The disconnect line is giving us grief. it iterates over the pins and removes them from the iterated structure.

The fix is to change this to:

pins = w.pins
while len(pins) > 0:
    p = pins[0]
    w.disconnect_pin(p)
    ew.connect_pin(p)

I'll push it up to the documentation_dallin branch. There are also some fixes to the verilog composer in that misnamed branch.

jacobdbrown4, 2021-07-09:
Awesome! I got what you pushed up to your branch and tried it out and it seems to have fixed it!

jacobdbrown4, 2021-07-12:
I found another issue that seems to be related to this same issue. In example netlist float_demo on branch dallin_documentation, the m_axis_result_tdata[0] net is disconnected after parsing it in.

jacobdbrown4, 2021-07-12:
I found a weird line in the original edif file that I think was messing up the parser.
                (net (rename m_axis_result_tdata_0_ "m_axis_result_tdata[0]") (joined
                (portref (member m_axis_result_tdata 63) (instanceref i_synth))
                (portref (member m_axis_result_tdata 63))
                )
                )
                (net (rename m_aXIS_result_tdata_0_ "m_axis_result_tdata[0]") (joined))
After removing the 'aXIS' line, it looks like it parsed it correctly. Not sure why that line is there, but after removing, it's all good.

thunder-hammer, 2021-07-13:
Oh interesting. the rename field is identical but the edif identifier is case sensitive. Do you know what program generated the EDIF? This looks more related to #116 I'm not sure this is an issue with the multi-bit handling.

jacobdbrown4, 2021-07-20:
The EDIF parser bug fix was released in 1.8.3

closed 2021-07-20 by jacobdbrown4