byuccl/spydrnet #182: Namespace manager bug

emonlux, 2022-07-01:
Unable to generate a TMR netlist
ValueError: Applying name would result in a naming conflict

namespace_manager -> init.py
Line 107 - if parent and parent in self.namespaces:

possible fix - if parent and element in self.namespaces:

emonlux, 2022-07-07:
Yeah I eventually figured that out. Parent and Parent seems redundant though

jacobdbrown4, 2022-07-07:
So it looks like it's an issue with upper-case and lower-case. The namespace manager does not take case into account when checking for duplicate names. So F1_reg_TMR_0_Q_VOTER and f1_reg_TMR_0_Q_VOTER are considered the same. It's interesting that this wasn't caught when the instance was created but rather when composing the netlist.

jacobdbrown4, 2022-07-07:
It's also interesting that your generated netlist had the same names just different case. Where did the netlist come from?

emonlux, 2022-07-07:
The netlist came from Vivado

jacobdbrown4, 2022-07-07:
> Yeah I eventually figured that out. Parent and Parent seems redundant though

As for this, it may seem redundant but I don't think it is. It's checking to make sure the element has a parent, and then it's seeing if it's in self.namespaces. The writer may have done this because 'None' (which the parent would be if it didn't exist) may be self.namespaces and may cause problems if we try to access it. So we if the parent is 'None' it is passed by.

jacobdbrown4, 2022-07-07:
That pull request looks like it gives the case sensitive option to the parser but not the composer. We probably need to add to it.

jacobdbrown4, 2023-04-24:
See the case_insensitive_naming branch for some work on this.

jacobdbrown4, 2023-04-24:
Note that this is related to issue #116

jacobdbrown4, 2023-10-26:
Note: the case_insensitive_naming branch contains code to tell the EDIF parser to be case sensitive for naming. I merged this code into next release but then removed it, so github thinks this branch has nothing new but it does.