byuccl/spydrnet #183: Issues parsing F4PGA-generated reverse-netlists.

JakeEdvenson, 2022-07-18:
When I use spydrnet on a reversed-netlist, I get an error like this: `AssertionError: expected . to begin parameter mapping but got ) Line: 8619` I determined that it is because the reversed netlist generated by F4PGA has several components like this: 
 ` (* KEEP, DONT_TOUCH, BEL = "CARRY4" *)
  CARRY4 #(
  ) CLBLL_L_X34Y123_SLICE_X51Y123_CARRY4 (
.CI(1'b0),
.CO({CLBLL_L_X34Y123_SLICE_X51Y123_D_CY, CLBLL_L_X34Y123_SLICE_X51Y123_C_CY, CLBLL_L_X34Y123_SLICE_X51Y123_B_CY, CLBLL_L_X34Y123_SLICE_X51Y123_A_CY}),
.CYINIT(1'b0),
.DI({CLBLL_L_X34Y123_SLICE_X51Y123_DO5, CLBLL_L_X34Y123_SLICE_X51Y123_CO5, CLBLL_L_X34Y123_SLICE_X51Y123_BO5, CLBLL_L_X34Y123_SLICE_X51Y123_AO5}),
.O({CLBLL_L_X34Y123_SLICE_X51Y123_D_XOR, CLBLL_L_X34Y123_SLICE_X51Y123_C_XOR, CLBLL_L_X34Y123_SLICE_X51Y123_B_XOR, CLBLL_L_X34Y123_SLICE_X51Y123_A_XOR}),
.S({CLBLL_L_X34Y123_SLICE_X51Y123_DO6, CLBLL_L_X34Y123_SLICE_X51Y123_CO6, CLBLL_L_X34Y123_SLICE_X51Y123_BO6, CLBLL_L_X34Y123_SLICE_X51Y123_AO6})
  ); `
Luckily I only need spydrnet to get the inputs and outputs so I cut out all of this information from every reverse-netlist I analyze, but this could be something to check out and possibly fix. Thank you!

jacobdbrown4, 2022-08-01:
I'll look into this. Can you attach a zip file of the netlist you were using?

jacobdbrown4, 2022-08-01:
I copied what you put in your comment above into a netlist and took a look. It was as I thought. I fixed it in the Verilog parser and it will be included in our next release (which should be soon).

closed 2022-11-09 by jacobdbrown4