byuccl/spydrnet #187: Strict code formatting

ganeshgore, 2022-08-06:
I can see several warning while using pylint or black during editing code.
- unused imports
- trailing whitespace spaces 
- long code lines 
- missing docstring

Is there any documentation detailing how to maintain consistency during code formatting?
I can see .pylintrc is provided by its not enforced during PR. Do you think it is a good idea to add that CI check?

I generally use Black for formatting because of its consistency, determinism, and speed.

jacobdbrown4, 2023-04-06:
Someone who used to work on SpydrNet once looked into making this an automated process. His work is on the pylint_jordi branch. We will look at making linting code a part of our process of doing things.

jordi1215, 2023-04-06:
Wasn't expecting to see my name after all these years. But I didn't get very far in using Pylint for the code base. So I am not sure if it is any good looking into the branch. 

We all have been using the Black formatter since I worked on this project 3 years ago. An easy fix for most of the problems @ganeshgore  mentioned could be potentially resolved by running the Black formatter on the whole repo. Might be a good idea to look into this. I think you can just run `black .` on the root directory, but I am not entirely sure @jacobdbrown4 

As for incorporating the pylint into the CI check, that should take a bit more work but should be straightforward, and is a very good idea if it hasn't been done already.

And for the documentation for code formatting. I vaguely remember that there should be one for spydrnet. But I can be wrong. Sorry for being lazy, I have since moved on to other projects and have deadlines to meet too. Good luck!

jacobdbrown4, 2023-04-06:
@wirthlin can you enable the pylint github actions to run?

jacobdbrown4, 2023-08-30:
Okay, the following is in the next-release branch and is waiting to be merged into master:

Pylint will now run as part of the github actions. Right now the threshold score is set at 7.5, so any final score below that will fail the run. This score can be increased in the future as our code formatting improves. The run currently passes, so the only way to make it not pass is if someone makes changes/additions that are formatted poorly and bring the score down.

closed 2023-09-14 by jacobdbrown4