byuccl/spydrnet #195: parse should work with Pathlib.Path

jgoeders, 2023-01-25:
`spydernet.parse` throws a `AttributeError: 'PosixPath' object has no attribute 'readable'` if you provide a PosixPath pathlib.Path object.

emonlux, 2023-02-08:
I believe that I was able to recreate the same error. Including .name at the end of the command (e.g. pathlib.Path('...').name) should fix the issue. Is this something that you want added into the parser in cases where the user does not include .name?

jgoeders, 2023-03-16:
Yes, .name works because it returns a string; however, the parse API should also accept a `pathlib.Path` object as that is the standard way of providing file paths in Python.

emonlux, 2023-03-30:
I added on the next release branch a flag to pass to the parser if pathlib.Path is being used. The flag is path_used
Does this fix the issue for you?

jacobdbrown4, 2023-04-01:
Instead of having to pass a parameter to parser, can we just check what the passed object is? If it's a string, treat like normal, and if it's a pathlib.Path object, get the filename string from it and continue on?

jgoeders, 2023-04-01:
I agree with @jacobdbrown4.  It's odd to have a flag.  It's expected that python libraries that accept a file path will just work with pathlib.Path objects.  

>  if it's a pathlib.Path object then get the filename string from it and continue on

Yes, you could add something like this to the start of the function:
```python
filename = str(filename)
```

However, you should really be keeping the path as a pathlib.Path object internally, and more correctly, if passed a string, convert it to pathlib.Path object right away.  

Right now it looks like the code is using os.path everywhere.  This is the old way of doing paths in Python.  The code should be using pathlib for all path handling and manipulations.  It's much more readable and the modern Python way of dealing with paths.

jacobdbrown4, 2023-04-05:
Thanks for the info. We should probably update SpyDrNet to use pathlib like you were saying.

emonlux, 2023-04-13:
@jacobdbrown4 and I have added this functionality to the parser on the next release branch. 

> > if it's a pathlib.Path object then get the filename string from it and continue on

Does this fix the issue for you? Also, I have started looking into updating the code from os.path to pathlib. I have noticed different types of paths e.g. PosixPath, PurePosixPath, PurePath, and Path as well as paths for windows. Is there a standard type of path that should be used?

emonlux, 2023-04-24:
# Remaining os.path in code

[spydrnet/compare/tests/test_compare_netlists.py](https://github.com/byuccl/spydrnet/blob/next_release/spydrnet/compare/tests/test_compare_netlists.py)
Line 17
Line 18

[spydrnet/__init__.py](https://github.com/byuccl/spydrnet/blob/next_release/spydrnet/__init__.py)
Line 121
Line 127
Line 133

[spydrnet/util/architecture.py](https://github.com/byuccl/spydrnet/blob/next_release/spydrnet/util/architecture.py)
Line 8
Line 10
Line 11
Line 12

[setup.py](https://github.com/byuccl/spydrnet/blob/next_release/setup.py)
Line 29
Line 30

[spydrnet/composers/verilog/tests/test_composer.py](https://github.com/byuccl/spydrnet/blob/next_release/spydrnet/composers/verilog/tests/test_composer.py)
Line 15
Line 80
Line 98

[docs/source/conf.py](https://github.com/byuccl/spydrnet/blob/next_release/docs/source/conf.py)
Line 20
Line 211

[spydrnet/parsers/edif/tests/test_edif_parser.py](https://github.com/byuccl/spydrnet/blob/next_release/spydrnet/parsers/edif/tests/test_edif_parser.py)
Line 56

[spydrnet/tests/test_example_netlist_functionality.py](https://github.com/byuccl/spydrnet/blob/next_release/spydrnet/tests/test_example_netlist_functionality.py)
Line 10

[spydrnet/parsers/verliog/tests/test_tokenizer.py](https://github.com/byuccl/spydrnet/blob/next_release/spydrnet/parsers/verilog/tests/test_tokenizer.py)
Line 13

[spydrnet/parsers/edif/tests/test_tokenizer.py](https://github.com/byuccl/spydrnet/blob/next_release/spydrnet/parsers/edif/tests/test_tokenizer.py)
Line 29

jacobdbrown4, 2023-05-10:
I have gone through and updated everything that uses os.path to use pathlib. I believe it is all working okay.

I might create some tests that use pathlib paths to make sure parsing and composing works okay with that.

- [ ] Create pytests that pass pathlib.Path objects to the parsers and composers rather than strings.

jacobdbrown4, 2023-05-15:
This is in next release branch and waiting to be released.

closed 2023-05-15 by jacobdbrown4