byuccl/spydrnet #196: Distinguishing pins in multi-bit ports

jgoeders, 2023-03-18:
If I have a multi-bit port, how do I distinguish between the various pins?

I have an 8-bit port called *Din*:

When I iterate over the pins they all look identical.  I would expect the *InnerPin* to have an *index* property, but I can't find one.  

```python
for pin in port.get_pins():
  print(pin)
```

```
<class 'spydrnet.ir.InnerPin'; connected to'<class 'spydrnet.ir.Wire'; Contained by Cable.name 'Din' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: False; lower index: 0>>'>
<class 'spydrnet.ir.InnerPin'; connected to'<class 'spydrnet.ir.Wire'; Contained by Cable.name 'Din' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: False; lower index: 0>>'>
<class 'spydrnet.ir.InnerPin'; connected to'<class 'spydrnet.ir.Wire'; Contained by Cable.name 'Din' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: False; lower index: 0>>'>
<class 'spydrnet.ir.InnerPin'; connected to'<class 'spydrnet.ir.Wire'; Contained by Cable.name 'Din' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: False; lower index: 0>>'>
<class 'spydrnet.ir.InnerPin'; connected to'<class 'spydrnet.ir.Wire'; Contained by Cable.name 'Din' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: False; lower index: 0>>'>
<class 'spydrnet.ir.InnerPin'; connected to'<class 'spydrnet.ir.Wire'; Contained by Cable.name 'Din' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: False; lower index: 0>>'>
<class 'spydrnet.ir.InnerPin'; connected to'<class 'spydrnet.ir.Wire'; Contained by Cable.name 'Din' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: False; lower index: 0>>'>
<class 'spydrnet.ir.InnerPin'; connected to'<class 'spydrnet.ir.Wire'; Contained by Cable.name 'Din' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: False; lower index: 0>>'>
```

Am I missing something?

ganeshgore, 2023-03-18:
You can get that using `port.pins.index(pin)` or `pin.port.pins.index(pin)` (if you don't have the relevant port in the scope)
Note: this just returns the unique index number in the list (not guaranteed to match with the Verilog index declaration)

emonlux, 2023-03-30:
Currently there is no function built in for this. I will look into adding this functionality and documenting it.

jacobdbrown4, 2023-04-26:
OuterPins do not have names, however their ports do. I'm guessing the 'name' in your proposed addition would be the port name and then the index would specify what pin. On the other hand, we can give OuterPins names. Perhaps they could be their port name + "_" + their index.

jacobdbrown4, 2023-10-31:
I have added an index property to InnerPin and OuterPin and updated \__str\__() to include index and port information.

I have also implemented the following as requested:

```
Instance.get_pin(name)
Instance.get_pin(name, index)
```

The default index is 0. If no pin matching the name and index is found, *None* will be returned.

These updates can be found on the next_release branch.