byuccl/spydrnet #197: Instance names have space at the end

jgoeders, 2023-03-31:
I am finding that instances in my netlist like this one:
```verilog
 CARRY4 \count_i_reg[12]_i_6 
       (.CI(\count_i_reg[8]_i_6_n_0 ),
        .CYINIT(\<const0> ),
        .DI({\<const0> ,\<const0> ,led_OBUF[14:13]}),
        .O(count_i0[15:13]),
        .S({\<const0> ,\count_i[12]_i_8_n_0 ,\count_i[12]_i_9_n_0 ,\count_i[12]_i_10_n_0 }));
```

end up with an instance name with a space at the end in spydernet.  For example, this evaluates to true:
```python
instance.name == r"\count_i_reg[12]_i_6 "
```

jgoeders, 2023-04-02:
Per <https://www.verilog.com/VerilogBNF.html#REF170>: 

>	An identifier is any sequence of letters, digits, dollar signs ($), and
	underscore (_) symbol, except that the first must be a letter or the
	underscore; the first character may not be a digit or $. Upper and lower case
	letters are considered to be different. Identifiers may be up to 1024
	characters long. Some Verilog-based tools do not recognize  identifier
	characters beyond the 1024th as a significant part of the identifier. Escaped
	identifiers start with the backslash character (\) and may include any
	printable ASCII character. An escaped identifier ends with white space. The
	leading backslash character is not considered to be part of the identifier.

This explicitly states that the backslash it not part of the identifier, but is a bit unclear on the whitespace character.

Per <https://www.verific.com/faq/index.php?title=Escaped_identifiers_in_RTL_files_and_in_Verific_data_structures>

> The escaping characters '\' and ' ' are not part of the name : 'foo' is the same object as '\foo '.

jacobdbrown4, 2023-04-06:
It would be simple to change the verilog parser to remove that space and then re-add it when composing. We already have an option to remove the space but maybe we should make it the default.

jacobdbrown4, 2023-04-11:
@jgoeders I updated the Verilog parser to remove the space in names and the composer to reinsert it during composing. I pushed the change to the next_release branch.

jacobdbrown4, 2023-04-24:
This has been merged into the main branch.

closed 2023-04-24 by jacobdbrown4