byuccl/spydrnet #198: Wrong pin indexing on partially connected output port

jgoeders, 2023-03-31:
I have a CARRY4 instance in my netlist, where a 3-bit signal is connected to a 4-bit output.  This netlist is generated by Vivado.

```verilog
 CARRY4 \count_i_reg[12]_i_6 
       (.CI(\count_i_reg[8]_i_6_n_0 ),
        .CYINIT(\<const0> ),
        .DI({\<const0> ,\<const0> ,led_OBUF[14:13]}),
        .O(count_i0[15:13]),
        .S({\<const0> ,\count_i[12]_i_8_n_0 ,\count_i[12]_i_9_n_0 ,\count_i[12]_i_10_n_0 }));
```

The issue is that the *count_i0* signal is being connected to pins O[3:1] instead of O[2:0].

For example, I iterate through pins on the instance, and print *pin port name*, *index*, and *wire* and get the following:
```
CI 0 <class 'spydrnet.ir.Wire'; Contained by Cable.name '\count_i_reg[8]_i_6_n_0 ' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: True; lower index: 0>>
CO 0 None
CO 1 None
CO 2 None
CO 3 None
CYINIT 0 <class 'spydrnet.ir.Wire'; Contained by Cable.name '\<const0> ' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: True; lower index: 0>>
DI 0 <class 'spydrnet.ir.Wire'; Contained by Cable.name 'led_OBUF' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: False; lower index: 0>>
DI 1 <class 'spydrnet.ir.Wire'; Contained by Cable.name 'led_OBUF' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: False; lower index: 0>>
DI 2 <class 'spydrnet.ir.Wire'; Contained by Cable.name '\<const0> ' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: True; lower index: 0>>
DI 3 <class 'spydrnet.ir.Wire'; Contained by Cable.name '\<const0> ' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: True; lower index: 0>>
O 0 None
O 1 <class 'spydrnet.ir.Wire'; Contained by Cable.name 'count_i0' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: False; lower index: 1>>
O 2 <class 'spydrnet.ir.Wire'; Contained by Cable.name 'count_i0' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: False; lower index: 1>>
O 3 <class 'spydrnet.ir.Wire'; Contained by Cable.name 'count_i0' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: False; lower index: 1>>
S 0 <class 'spydrnet.ir.Wire'; Contained by Cable.name '\count_i[12]_i_10_n_0 ' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: True; lower index: 0>>
S 1 <class 'spydrnet.ir.Wire'; Contained by Cable.name '\count_i[12]_i_9_n_0 ' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: True; lower index: 0>>
S 2 <class 'spydrnet.ir.Wire'; Contained by Cable.name '\count_i[12]_i_8_n_0 ' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: True; lower index: 0>>
S 3 <class 'spydrnet.ir.Wire'; Contained by Cable.name '\<const0> ' <class 'spydrnet.ir.Cable'; is_downto: True; is_scalar: True; lower index: 0>>
```

jacobdbrown4, 2023-04-01:
@jgoeders can you share the netlist you were using so I can take a look?

jacobdbrown4, 2023-04-01:
I was able to use one of the built-in spydrnet example netlists to look at partially connected ports on instances. I saw the same thing you're seeing, that is, partially connected ports are connected on the upper side rather than the lower side. 

After looking at an older version of spydrnet and not seeing this issue, I now think this is a bug introduced last summer when I made several changes to the verilog parser. 

I have a fix that seems to work. I am going to make sure that this fix doesn't break anything else and then I will push it up to the next-release branch.

jgoeders, 2023-04-03:
@jacobdbrown4 OK, thanks.  Could you let me know once the fix is pushed up to a branch that I can try out?  Thanks!

jacobdbrown4, 2023-04-05:
@jgoeders  I just pushed the fix to the next_release branch. Let me know how it goes.

jgoeders, 2023-04-12:
This is working now in `next_release`.  Thanks.

closed 2023-04-12 by jgoeders