byuccl/spydrnet #201: [verilog] Primitives

agg23, 2023-04-02:
Unsure if this actually applies to the project, as it probably doesn't come up in most (any?) netlists, but Verilog has the concept of primitives, both builtin (`XOR`, `AND`, etc), and [user generated](https://www.chipverify.com/verilog/verilog-udp).

Since primitives are created using solely a truth table, they can be instantiated without giving them a name:
```verilog
some_module inst1 (.in(0), .out(test), ...);
and (a, b, q);
```

The Intel simulation libraries use these concepts throughout, and thus are not currently parsible. Again, not sure if this applies, since these files are full simulation-ready implementations, not barebones netlists.

jacobdbrown4, 2023-04-05:
Our goal has been to parse structural Verilog netlists, so we have never needed support for this. We may look at adding support for this sometime though.

jacobdbrown4, 2023-10-06:
If I understand correctly, the desired functionality is not having to give instances a name. Is that it? @agg23

agg23, 2023-10-11:
Sorry, somehow I wasn't notified for this mention. The desired functionality is to properly parse the Intel simulation libraries, which includes the use of the primitive directive:

```verilog
primitive and (input a, input b, output out);
    ...
endprimitive
```

Since these represent _primitives_, they are allowed to be instantiated without a name, as there's no point in having `and_wire1_wire2` repeated across many different instantiations. SpyDrNet should be able to parse those primitives without throwing an error.

jacobdbrown4, 2023-10-19:
@agg23 This looks like something we could support. If the Verilog parser sees that the cell type is a primitive, it will not look for a name. What primitives should be supported? Can you provide a list of their names and ports?