byuccl/spydrnet #206: Move Tests Out of Main Repo

jacobdbrown4, 2023-04-24:
This comes from a suggestion from @jgoeders and @agg23 . Test code should not be included in the main repo but rather be at an external location. See issue #200.

agg23, 2023-04-24:
You almost definitely don't want to move the test code out of the _repo_, just out of the package. So when I do `pip install spydrnet`, I don't want to download the test code, but when I do `git clone [repo URL]`, I do want to get the code (admittedly you could use a Git submodule, but you don't want to do that).

jacobdbrown4, 2023-04-24:
Thanks for the clarification. I'll edit the issue's title.

jacobdbrown4, 2023-04-26:
I have created a branch named 'move_tests' and have begun work on this.

jacobdbrown4, 2023-05-10:
I have moved the tests out of the package on the branch called 'move_tests'. All seems well so I expect to merge this into the main branch sometime.

jacobdbrown4, 2023-05-15:
I will merge this into next release and then we will wait to release it later this summer.

closed 2023-05-15 by jacobdbrown4