byuccl/spydrnet #213: Getting recursion error when parsing a netlist

arun-chaubey, 2023-10-05:
Hi,

When trying to load up a verilog gate netlist written out of Synopsys synthesis tool, I am getting `RecursionError: maximum recursion depth exceeded in comparison`

Below is the code and detailed error trace:

```
import spydrnet as sdn
netlist = sdn.parse('Documents/hello_world.v')
```

```
Traceback (most recent call last):
  File "Desktop/spydrnet_hello_world.py", line 3, in <module>
    netlist = sdn.parse('Documents/hello_world.v')
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/parsers/__init__.py", line 61, in parse
    return _parse(filename, architecture)
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/parsers/__init__.py", line 79, in _parse
    parser.parse()
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/parsers/verilog/parser.py", line 114, in parse
    self.parse_verilog()
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/parsers/verilog/parser.py", line 210, in parse_verilog
    self.parse_module()
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/parsers/verilog/parser.py", line 344, in parse_module
    self.parse_module_body()
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/parsers/verilog/parser.py", line 610, in parse_module_body
    self.parse_cable_declaration(params)
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/parsers/verilog/parser.py", line 742, in parse_cable_declaration
    self.parse_cable_declaration({}, var_type)
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/parsers/verilog/parser.py", line 742, in parse_cable_declaration
    self.parse_cable_declaration({}, var_type)
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/parsers/verilog/parser.py", line 742, in parse_cable_declaration
    self.parse_cable_declaration({}, var_type)
  [Previous line repeated 976 more times]
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/parsers/verilog/parser.py", line 735, in parse_cable_declaration
    cable = self.create_or_update_cable(
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/parsers/verilog/parser.py", line 1400, in create_or_update_cable
    cable = self.current_definition.create_cable()
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/ir/definition.py", line 388, in create_cable
    cable = Cable(name, properties, is_downto, is_scalar, lower_index)
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/ir/cable.py", line 43, in __init__
    _call_create_cable(self)
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/global_state/global_callback.py", line 52, in _call_create_cable
    func(*args, **kwargs)
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/plugins/namespace_manager/__init__.py", line 54, in create_cable
    cable[".NS"] = self.default
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/ir/first_class_element.py", line 76, in __setitem__
    global_callback._call_dictionary_set(self, key, value)
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/global_state/global_callback.py", line 152, in _call_dictionary_set
    func(*args, **kwargs)
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/plugins/namespace_manager/__init__.py", line 110, in dictionary_set
    self.apply_namespace(value, target_namespace, element)
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/plugins/namespace_manager/__init__.py", line 230, in apply_namespace
    element[".NS"] = value
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/ir/first_class_element.py", line 76, in __setitem__
    global_callback._call_dictionary_set(self, key, value)
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/global_state/global_callback.py", line 152, in _call_dictionary_set
    func(*args, **kwargs)
  File "/opt/anaconda3/lib/python3.9/site-packages/spydrnet/plugins/namespace_manager/__init__.py", line 90, in dictionary_set
    if key == ".NS":
RecursionError: maximum recursion depth exceeded in comparison
```

Has anyone else seen this issue before or can please let me know if I might be missing something on my end ?

Thanks & Regards,
Arun

jacobdbrown4, 2023-10-05:
Hi @arun-chaubey, are you able to share the netlist you're trying to parse so I can try it out myself and see what's going wrong?

arun-chaubey, 2023-10-05:
@jacobdbrown4 , thanks for your prompt response. Unfortunately the netlist is a key IP so wouldn't be able to share :(

jacobdbrown4, 2023-10-05:
That's okay. It looks like line 742 in the verilog parser is being called over and over again. This happens when it parses multiple wire declarations on one line, like the following:

`wire a, b, c, d;`

parse_cable_declaration() parses a wire declaration and if it sees that there is a comma instead of a semicolon, it will call itself again to parse the next wire. The problem with this is that it is implemented in a recursive fashion, not iterative. My guess is that there is a large amount of wires declared in a single line, so parse_cable_declaration() calls itself over and over until the max recursion depth is reached. 

I will work on fixing this. In the meantime, you can try looking at the netlist and breaking up the wire declarations into a multiple lines to see if that helps.

jacobdbrown4, 2023-10-05:
I went ahead and made my own dummy netlist to test my theory and I got the same error (max recursion depth exceeded from line 742). Fixing it was simple, and the fix can be found on the next_release branch. I'm not sure when it will be merged into the master branch.

arun-chaubey, 2023-10-06:
Awesome, thanks @jacobdbrown4

arun-chaubey, 2023-10-06:
Btw @jacobdbrown4 if you can provide the fix so I can put it in locally to proceed in the meantime, would be very helpful. Thanks

jacobdbrown4, 2023-10-06:
You should be able to access the 'next_release' branch. The fix is found [here](https://github.com/byuccl/spydrnet/blob/dbc83ca3d8d985510532bb708c7d90d9ae3c3351/spydrnet/parsers/verilog/parser.py#L718) (line 718 to 753). I think the easiest way to start using the fix is to clone the repo, switch to the next_release branch, and install spydrnet from there.