byuccl/spydrnet #216: `\<const0>` and `<const1>` missing drivers

jgoeders, 2023-10-19:
The netlist generated by spydrnet uses `\<const0>` and `<const1>`  for constant 0 and 1, but it doesn't actually assign or drive these signals.

For example, I have the following netlist:

```verilog
module test (
    input a,
    input b,
    output o
);

  LUT6_2 #(
    .INIT(64'h0000000000000000)
  ) MY_LUT (
.I0(a),
.I1(b),
.I2(1'b0),
.I3(1'b0),
.I4(1'b1),
.I5(1'b1),
.O5(),
.O6(o)
  );

endmodule
```

I read in and write this out using spydrnet:

```python
import spydrnet as sdn

ir = sdn.parse("netlist.v")

sdn.compose(ir, "netlist_out.v", write_blackbox=False)
```

Which produces the following output file:
```verilog
module test
(
    a,
    b,
    o
);

    input a;
    input b;
    output o;

    wire \<const1> ;
    wire \<const0> ;
    wire o;
    wire b;
    wire a;

    LUT6_2 #(
        .INIT(64'h0000000000000000)
    )
    MY_LUT
    (
        .I0(a),
        .I1(b),
        .I2(\<const0> ),
        .I3(\<const0> ),
        .I4(\<const1> ),
        .I5(\<const1> ),
        .O5(),
        .O6(o)
    );
endmodule
```

However, the `\<const0>` and `<const1>` nets have no driver.  

In netlists generated by Vivado, they have drivers for these signals that look like this:
```
  GND GND
       (.G(\<const0> ));
  VCC VCC
       (.P(\<const1> ));
```

One solution would be for spydrnet to also create drivers for these signals.  If you take this approach, perhaps it should be in a tool-agnostic fashion, like this:

```verilog
assign \<const0> = 1'b0;
assign \<const1> = 1'b1;
```

This unfortunately introduces `assign` statements into the Verilog, which I know spydrnet supports, but in my opinion makes things a bit messy.  

I would probably prefer just using the `1'b1` and `1'b0` literals in place of  `\<const0>` and `<const1>`  (for example, this is the approach taken by the netlists generated by fasm2bels)

jacobdbrown4, 2023-10-23:
I have implemented a fix for this on the next_release branch. If the \\<const0> or \\<const1> wire is driven by an instance (for example GND or VCC), the name is composed out. Otherwise it is changed to 1'b0 or 1'b1. Can you try it out @jgoeders and let me know what you think?

jgoeders, 2023-10-26:
@jacobdbrown4 Yes, thanks.  It seems to be working well.

closed 2023-10-30 by jacobdbrown4